AstroPrint/OctoPrint-AstroPrint #8: Not seeing any way to link Astroprint account to Octopi

kennethpangky, 2017-12-16:
https://imgur.com/a/5ExA2

As in image, there is no way for me to enter the Access Key from Astro Cloud Settings into the Astroprint plugin. I have tried enabling CORS but no change. Or am I doing something wrong?

CoDanny, 2017-12-16:
When you install the plugin, a new tab called “AstroPrint” is added to the OctoPrint menu ( alongside “Control”, “Terminal”, “Timelapse”, etc)

It’s there that you make the connection.

This seems to be confusing, you’re not the first one to ask. We’ll add an option in settings

closed 2017-12-16 by CoDanny

BenWoodford, 2017-12-20:
Yeah the confusion is purely down to the fact that you've got the tab on the dashboard for adding the key, but then every other setting is under settings.

IMO, the dashboard tab is unnecessary, the settings tab is all you need.