AstroPrint/OctoPrint-AstroPrint #10: Astroprint tab doesn't follow CSS for the rest of Octoprint

BenWoodford, 2017-12-20:
This is mostly an issue if you have a dark theme installed. Headings, pagination, the main bar at the top of the tab should all be following the existing colours wherever possible to avoid mismatches when themes are used.

CoDanny, 2017-12-28:
@BenWoodford Could you give more info regarding how to follow themes? Is there documentation?

CoDanny, 2018-01-09:
Closing as there has been no follow up

closed 2018-01-09 by CoDanny

BenWoodford, 2018-01-10:
Sorry, totally forgot about this. There's no CSS guide, but it's the Themeify plugin in use. Everything else follows the correct colours but Astroprint doesn't. See below

[image]

CoDanny, 2018-01-10:
@BenWoodford Yes, we saw that. Note that the whole tab is our UI and we were able to match the style in everything except that header... At the moment I don't know how to get that container to "behave".