AstroPrint/OctoPrint-AstroPrint #13: [Request] Add support to third-party plugins

leobel96, 2017-12-24:
Please add support to octoprint third-party plugins such as Octoprint-enclosure (https://github.com/vitormhenrique/OctoPrint-Enclosure) so that printer can also be switched on and off from everywhere.

leobel96, 2019-08-11:
@Alfonsopl any news?

CoDanny, 2019-08-12:
Hi @leobel96, while we feel this is useful and at some point we would like to do it. We're not in a position to take on this work now as our small team is focused on other more pressing things.

AfromD, 2020-01-13:
@Alfonsopl I have a similar request.
I use Octoprint with the PSUControl plugin. This enables my Raspi to control power to the printer. 

Basically, it is all automatic, nothing to do for Astroprint. There is just one big problem: When Octoprint has no connection to the printer (because the printer has no power), Astroprint will not start a print job.

I have looked at the code, but it was not instantly obvious to me how it works. _What do I need to change so that Octoprint signals Astroprint that it is ready to start printing, even when the connection is down?_ Because with my setup, it is. So I would simply like the plugin to only signal "not ready" when it is busy, but not when the connection between octoprint and the printer is down.

AfromD, 2020-01-23:
Ok, I HACKED this as follows (only recommended if you know what you are doing)

caveat: this has no user interface for configuration, and uniquely works for my setup, where GPIO pin 4 controls a relay which controls printer power. So it does not interface with PSUcontrol, but operates the relay independently. Works for me, might need to work differently for others.

To make sure that Astroprint will send files to a printer that is not powered (octoprint running of course) add the following change to 
home/pi/oprint/lib/python2.7/site-packages/octoprint_astroprint/boxrouter/handlers/requesthandler.py
add 
import RPi.GPIO as GPIO
at the top

in
class RequestHandler(object):
def initial_state(self, data, clientId, done):
change line
    'operational': self._printer.is_operational(),
to
    'operational': self._printer.is_operational() or not(GPIO.input(4)),

Then, to make sure the printer is switched on after gcode is successfully downloaded, and to wait for the printer to become available, change file 
/home/pi/oprint/lib/python2.7/site-packages/octoprint_astroprint/AstroprintCloud.py
add 
import RPi.GPIO as GPIO
at the top

in
class AstroprintCloud():
def printFileIsDownloaded(self, printFile):
add
		GPIO.setup(4, GPIO.OUT)
		GPIO.output(4, 1)
		
		while not(self._printer.is_operational()):
			time.sleep(1)
above
		if self._printer.is_printing():

As python reads formatting, be careful to use accurate indents with tab stops.
All these changes will probably be overwritten every time the plugin updates.