AstroPrint/OctoPrint-AstroPrint #16: Administrator Access?

3Dude10, 2018-01-05:
Trying to get the AstroPrint Plug-in to work but I am confused. It says “only accounts with administrator access can interact with the AstroPrint Plugin.” Anybody know what I did wrong and a way to get it working?

CoDanny, 2018-01-05:
You need to have admin rights in your OctoPrint account to enable this plugin

3Dude10, 2018-01-06:
Can I get these rights, is it just in the settings somewhere? I am fairly new to OctoPrint, sorry if this is a dumb question.

CoDanny, 2018-01-06:
If the box is yours, you should log into OctoPrint with the username that you used when you first setup the box.

3Dude10, 2018-01-06:
Just want to make it clear, this is the Raspberry Pi 3 running Octoprint not an AstroBox. I am logged in on OctoPrint But I cant use the AstroPrint because it says the message above.

3Dude10, 2018-01-06:
[https://plus.google.com/102577967477565381385/posts/LpC2iX4HNPn](url) Here is the link to my orginal question on google communities.  It has a picture of my screen too.

CoDanny, 2018-01-06:
@3Dude10 I understand. My answer was related to you octoprint use, not the atroprint one.

3Dude10, 2018-01-06:
Sorry, I misunderstood you. I’ll check it right now.

3Dude10, 2018-01-06:
It is working now, thanks for all your help and staying with me.

CoDanny, 2018-01-06:
No worries. Glad it's working :)

GingerSneak, 2018-01-09:
This issue appears for me as well. I have only one user in my OctoPrint install. I've verified that my user has admin rights in the YAML users file. Can this be reopened?

CoDanny, 2018-01-09:
@GingerSneak We're going to investigate. Note that there we have over 1000 users using the plugin successfully. There has to be something in your (and a few others) setup. Can you think of anything that might be different? What platform are you on?

GingerSneak, 2018-01-09:
@CoDanny, details below:

OctoPrint version: 1.3.4
Plugin version: 1.0.1
OctoPi version: 0.14.0
Hardware: Raspberry Pi Zero W

ihateu3, 2018-01-10:
Same issue here as well...

CoDanny, 2018-01-11:
@ihateu3 Which Octoprint update, what version are you running and which version was working?

GingerSneak, 2018-01-12:
@CoDanny, is there a log file I can look at or debugging that can be enabled to give more information?

ihateu3, 2018-01-12:
I am not really sure which version that I was on before, but I only have had octoprint for around a month, so it was current when I built it, I am currently on 1.3.6

As far as any logs, I didn't really find anything of value, but then again I do not know really what I am looking for.  I attached a log for you to see.
[octoprint.log.2018-01-07.txt](https://github.com/AstroPrint/OctoPrint-AstroPrint/files/1624653/octoprint.log.2018-01-07.txt)

CoDanny, 2018-01-12:
@GingerSneak In the logs directory there should be a log specific to the plugin, but it might not give any information since you can't even connect it due to the non-admin account. You can also look for javascript errors in your browser's web console.

@ihateu3 Your logs actually shows some exceptions that our plugin is putting out. It's helpful. you do have the plugin connected, correct?

Alfonsopl, 2018-01-12:
In the last commit that was made to adapt the plugin to support browsers in "strict mode" a variable lost the scope and was not able to recognize the user's login.
In the next version will be fixed, meanwhile you can login in Octoprint with "remember me" selected and reload the page to access the Astroprint tab

GingerSneak, 2018-01-12:
Just tried, no luck on the 'Remember Me' checkbox, it still responds the same.

PDunham113, 2018-01-13:
@GingerSneak I had the same issue this morning, but it dissapeared after following Alfonsolpl's advice. I know it's a silly question, but did you refresh the page after logging in with the 'Remember Me' checkbox?

The Astroprint plugin DID NOT work when I re-logged in. It only started working after I refreshed the page.

GingerSneak, 2018-01-13:
@PDunham113, just retried with a different machine and browser. Still no joy. :-(.
I think I'm going to try to roll back a few revisions and see if that helps.

GingerSneak, 2018-01-15:
Rolled back to 1.0.0, resolved the issue. When 1.0.2 is released, I'll be happy to install and test.

On a side note, thanks to @CoDanny and @Alfonsopl for making/contributing to this plugin, and for your responsiveness and diligence with this issue. Is there somewhere I can donate?

CoDanny, 2018-01-15:
@GingerSneak Thanks for the offer to donate. You can donate here: https://www.paypal.me/3DaGoGo

closed 2018-01-17 by CoDanny

nicksears, 2018-06-01:
have you tried refreshing the page? That's what did it for me.

commit 0911b199 (2018-01-17): Fixes #16