AstroPrint/OctoPrint-AstroPrint #19: Plugin keeps forgetting my key

BenWoodford, 2018-01-12:
Or something like that. I regularly find that Astroprint no longer knows about my printer, and I have to re-enter my Astroprint key.

Any ideas?

CoDanny, 2018-01-12:
It's likely that the access token is being deleted. How often does this happen? Can you give more details?

Do you log out of your OctoPi? Does it just happen by itself?

GitSoks, 2018-01-14:
I have the same problem my key is reseting every time I disconnect the raspberry from power (i don't know if this is also happens if I shut it down normally)

foo25amc, 2018-02-26:
This also happens to me every time I shutdown or reboot.

Alfonsopl, 2018-02-26:
Could you provide more information about your environment and octopi, and plugin version?

foo25amc, 2018-02-26:
I'm using a RPi3 running OctoPrint v1.3.6, OctoPi v0.14.0, and Astroprint v1.0.2. All packages are also up to date.

Alfonsopl, 2018-09-03:
I can not reproduce the error.
Could you provide more information about your environment and octopi, and plugin version?
Also are you logged to your octopi account after you reboot/shutdown?

Joe-houghton, 2018-09-05:
Hi,
Sorry about the delay.
I am using Octopi 1.3.9 (although it was happening in 1.3.8 too)
I installed AstroPrint plugin via the plugin manager and it says version AstroPrint (1.1.0).

Is there any more information that I can give you?

Joe

raffaelps, 2018-09-14:
I had the same error, every time when my Octopi shutdown, when I turn on the Octopi the plugin lost the key.

- Octopi Version 1.3.9
- AstroPrint Version 1.1.0

Thanks

Alfonsopl, 2018-09-14:
I have try with different versions of plugin and octopi and I am still not able to reproduce this error.

I think that the error could be that the plugin is not saving data into its database, that is why it can not find your access token.

Could you check if there is any encrypted data (or even file created) on /.octoprint/data/astroprint/octoprint_astroprint.db ?

raffaelps, 2018-09-19:
No @Alfonsopl, there isn't any encrypted data.

jmichelsen, 2018-09-19:
I experience this issue too. I thought I had a corrupted SD card so I actually just rebuilt my octopi setup on a new card with the current version of octopi and astroprint. If the pi gets rebooted, regardless of the method (shutdown, restart, loss of power), I have to re-enter my API key.

CoDanny, 2018-09-19:
@raffaelps  if this file ```/.octoprint/data/astroprint/octoprint_astroprint.db``` doesn't exists, that is the problem. Now we need to find out why it can't be created. Are you able to access the plugin logs and send them to us?

FedorSosnin, 2018-09-25:
Sadly, having the same issue.

daijoubu, 2018-09-26:
Myself as well.  I did find the settings file specified on my raspberry pi,
but can't confirm if the key was correctly stored or not.

Selfsustained, 2018-10-09:
@CoDanny I am not finding any astroprint directories after install. Using both putty and WinSCP over ssh to rpi3      Octoprint version: 1.3.9
                       Octopi version: 0.15.1
                       Astroprint version: 1.1.0

moalemadi, 2018-10-29:
Did any one find the solution to this problem? I have fully setup the printer to run remotely from powering on/off, cam and Octopi. Now it is useless since I have to be connected to the local network to enter the key every time.

ckohljr, 2019-01-15:
Bump - I also have this issue when I power cycle my OctoPI.  I have recently installed OctoPrint image and AstroPrint plugin within the last 2 weeks and it loses my key every time I reboot. Any Help?

Alfonsopl, 2019-01-15:
> Bump - I also have this issue when I power cycle my OctoPI. I have recently installed OctoPrint image and AstroPrint plugin within the last 2 weeks and it loses my key every time I reboot. Any Help?

Could you report what permissions has /.octoprint/data/astroprint/octoprint_astroprint.db ? Looks like the data is not being stored on database, and that is why the plugin "loses" the key on reboot. Some of the users has already reported that this file is empty.

ckohljr, 2019-01-16:
Thanks for the quick response, here are the permissions on the db:

pi@ender3octopi:~/.octoprint/data/astroprint $ ls -l
total 16
-rw-r--r-- 1 pi pi 16384 Jan 16 05:00 octoprint_astroprint.db
pi@ender3octopi:~/.octoprint/data/astroprint $

flyingelephant19, 2019-02-21:
i get the same issue. well everytime i shutdown the PI and restart, the creditials are gone and have to re enter and on one of my PI's i am running the latest 0.16 octopi.

markarsenich, 2019-02-23:
I'm also having the same issue and I have NO IDEA how to fix it... every time I reboot I loose the access key. Upon print completion my Octroprint sends a shutdown request via IFTTT to my wifi plug and it turns everything off thats plugged into my surge bar. when i turn everything back on theres no key, i have to log into astroprint.com, get the key, copy it over. has essentially made Astroprint (which i love) useless to me. PLEASE HELP!

CoDanny, 2019-02-23:
@markarsenich Does the same happen if you don't turn everything off that way and instead simply reboot the box via the octoprint options?

markarsenich, 2019-02-24:
FIXED! All I did was uninstall reboot, re-install reboot, now it works, cut power multiple times, stays logged in.

flyingelephant19, 2019-02-25:
mark, i cant say the same for me :(

flyingelephant19, 2019-02-25:
are there any logs that can be given ?

markarsenich, 2019-02-26:
well spoke too soon, not fixed, logging out on power off/on again... about to uninstall this plugin

drzejkopf, 2019-03-18:
I experience very same problem with plugin, and it's really irritating. If it's really related to other plugins then maybe it's good idea for everyone experiencing it to post which (non-bundled) plugins they have installed, then it will be easier to tell which plugin/plugins may possibly cause problems, I will start:
-Action Commands (0.1)
-AstroPrint (1.2.0)
-DeleteAfterPrint Plugin (1.2.0)
-Exclude Region (0.1.3)
-Fan Speed Control (0.2.1)
-Filament Manager (0.5.3)
-FileManager (0.1.3)
-Firmware Updater (1.1.0)
-GcodeEditor (0.2.6)
-ipOnConnect (0.2.1)
-M73-Progress-Plugin (0.1.0)
-OctoKlipper (0.2.5)
-PrintTimeGenius Plugin (1.3)
-Tempsgraph Plugin (0.3.3)
-Themeify (1.2.0)

CoDanny, 2019-03-18:
Is it possible to disable/delete the "DeleteAfterPrint Plugin" to see if that fixes the issue?

Steveps3, 2019-03-27:
I recently (2 months or so) changed my printer setup so that my pi is powered directly from the printer. Since i did this I am also losing my Astroprint API setting. This only happens once a week or so. Which is quite odd. Obviously the API is being written to the memory card but something is occasionally deleting it. I am on the verge of ditching Astroprint because is becoming so annoying.

CoDanny, 2019-03-27:
@Steveps3 since it was working before your change, why not looking at what changed that makes the plugin behave like this? Then we can maybe fix it.

Steveps3, 2019-03-27:
I thought that I'd explained what had changed? My raspberry pi is now powered from the printer. This means that when the printer turns off, so does the pi. There is something in the octoprint software that is just losing the astroprint API setting. This only happens maybe once a week so the problem is not as described above. The file that stores the API key is there most of the time.

CoDanny, 2019-03-27:
Ok. So the problem here is that the PI is being cut off without a graceful OctoPrint shutdown? I don't think there's much we can do about this. The system will be caught off guard by a sudden loss of power. In some cases the file with the data might not have been written to the system. This is the reason OctoPrint has a software shutdown/halt that should be used before cutting power.

Steveps3, 2019-03-27:
I have tried using the shutdown and it does the same thing. Which is why I don't bother with it any longer. Also why would cutting the power delete the API setting. The API setting can be there for 4-5 power cycles and then it is gone. It just doesn't make sense. Maybe I could change the permissions on the file so that it cannot be written to once it has the API setting.

CoDanny, 2019-03-27:
The same file is used to store other settings so it would depend of what the machine is doing at the time the power is cut. We'll keep looking into it. I thought this could've bee a solid lead but it appears not. We are completely UNABLE to reproduce this behavior and I would say it happens to less than 5% of the plugin users.

It's quite frustrating for us too as the first step to fix a problem is to be able to reproduce it in a controlled environment. Anyway thanks for your patience, we will eventually find this 🐛 and fix it.

Steveps3, 2019-03-27:
I am a software engineer so I totally understand your frustration. Being able to recreate a problem is half the battle. I have just spent the last 2 days trying to recreate something. I asked someone about it and they were able to show me the correct settings to recreate it. So frustrating :)

Maybe I will try the uninstall / reinstall method but I do have 2 printers and they both exhibit the same behaviour. I have considered going back to having the pi powered all the time but that has the complication that when I do turn on the printer, Octoprint doesn't automatically connect to it. So i have to "connect" manually. So I have a toss up between having to connect manually every time or having to reentered the API key once a week.

CoDanny, 2019-03-28:
There's a new version that add some logging. Please update to it and when this error happens again, reach out so we can take a look at your log.

Steveps3, 2019-04-02:
Well it's done it again. So if you want to tell me where to find the log ans where to send it then I'll be glad to do so.

CoDanny, 2019-04-02:
You can access and download files here. If you don't mind [open a ticket with us](https://astroprint.zendesk.com/hc/en-us/requests/new) to share logs. We will post back here when something relevant is found.

[image]

zigzaggin, 2019-04-14:
I had this issue as well. 

I was able to correct the issue by adjusting the permissions on the octoprint_astroprint.db file located in ~/.octoprint/data/astroprint/

I pushed the settings out to 777 because frankly its a closed box and I don't really care... but lower may also work.

Prior to the changed permissions, pulling the power cord caused the plugin to lose my API key, and now I am retaining it after multiple hard power resets.

Command to run if you're newer to pi, but please understand chmoding something to 777 isn't great practice, and starting lower and working up may be worthwhile.

sudo chmod 777 ~/.octoprint/data/astroprint/octoprint_astroprint.db

CoDanny, 2019-05-06:
Could you try installing the new version of the plugin (1.3.0) and see if this problem goes away. You might have to enter the credentials one more time after installation.

Steveps3, 2019-05-06:
Ok, I’ll try that. 
Thanks

Steveps3, 2019-05-10:
It failed after just a day. Same as before, astroprint cannot find the printer.

CoDanny, 2019-05-10:
@Steveps3 I'm a bit confused but this sentence "astroprint cannot find the printer". Do you mean it disconnects from the cloud? Does it ask you for the key again when you go to your OctoPrint interface?

what do you have in the ```/.octoprint/data/astroprint``` directory?

Steveps3, 2019-05-10:
No, it simply says "no remote connections"

CoDanny, 2019-05-10:
Ok, what does the OctoPrint interface say? and also if you could look in the directory I mentioned, that would help

Steveps3, 2019-05-10:
I am not at home right now. Which is why I need astroprint to be working. I will check tonight.

Steveps3, 2019-05-10:
That directory doesn't exist. Which might explain a few things.

sloth711, 2019-05-11:
hello

i have the same problem with the lastest version on octoprint and astroprint

after i discconnect my print 1h of plug on app android says " you need be logged"

to solve i need to re-install every time new key

please solve that 

thank you

rchoughton, 2019-05-15:
Has anyone figured out how to solve this problem?

CoDanny, 2019-05-15:
@rchoughton Our best guess right is that in some installations filesystem permissions are preventing our files from being written and there not able to persist the key values.

rchoughton, 2019-05-16:
Are these permissions that I can go in and change? Will they will stick making the system remember the key?

CoDanny, 2019-05-17:
You can change the permissions of the directory: ```/home/pi/.octoprint/data/astroprint```

@Steveps3, the path I gave you before was wrong. Can you check this new one?

Steveps3, 2019-05-18:
I have two files in that directory
print_files.yaml 
user.yaml

Steveps3, 2019-05-18:
These are the permissions that I have
[image]

CoDanny, 2019-05-18:
Thanks @Steveps3, What about the permissions of the files themselves? Also does the octoprint daemon run under the 'pi' user?

Steveps3, 2019-05-18:
The file permissions are
[image]

Yes, everything is running under the pi user.

CoDanny, 2019-05-18:
can you give write permissions to all and reboot just to test if that helps with your problem. You might have to enter the key one time after reboot and then it should stick

Steveps3, 2019-05-18:
Done. I will have to see how it gets on.

sloth711, 2019-05-18:
Hello... The problem it's solve?

Steveps3, 2019-05-18:
It is a bit optimistic to say that.

[15 more replies]

commit bc70ca00 (2018-01-16): Token doesn't refresh on background anymore but when it is necessary Fixes #19
closed 2019-08-11 by CoDanny