AstroPrint/OctoPrint-AstroPrint #21: Robo Integration questions

AllenMcAfee, 2018-01-19:
Hey Astro team!  I'm testing out the plugin for the R2 and C2 printers, and have two issues I ran into:

1.  Camera:  We are showing the "No Camera Detected" message when connected to the service.  We are using the standard Octoprint camera feature, and I have verified the camera is indeed turned on.  The camera feed path we are using is "/webcam/?action=stream" and CORS is indeed activated.  They are displaying correctly on our own mobile apps, but not on the Astro cloud and mobile apps.  Any thoughts?

2. Control:  Same issue as above.  When connected to the same local area network, we are showing the "Knock Knock" message.  Once again CORS is enabled and the other features are working correctly.

Is there a config.yaml schema you are specifically looking for?  Or maybe a specific version of Octoprint this is setup for?

AllenMcAfee, 2018-01-19:
As a note, the camera issue is also present on AstroCloud.  As of now there is no camera feed available.

AllenMcAfee, 2018-01-19:
Update:  From the Octoprint dashboard, I just noticed the "Search for Camera" option when I clicked on my user name.  I probably only found this because I was hunting for it.  It is my suggestion that the path be discovered automatically, or the "Search for Camera" button be brought to the front end a little better.  

So Robo is clear on marketing, the camera view only supports still images, correct?

The Controls issue described above is still present when connected to a local network.

CoDanny, 2018-01-20:
@AllenMcAfee The next version of our mobile app will have better support for octoprint instances running this plugin. The current live version is still missing some things.

AllenMcAfee, 2018-01-30:
@CoDanny just to note: this is existing on both AstroCloud and iOS.  Just mentioning this per your response about the mobile app update, but not about the cloud issue.

CoDanny, 2018-02-01:
Hi @AllenMcAfee,

1) I'll look into this more deeply

2) This is expected as the mobile app doesn't yet fully support local OctroPrint and the cloud UI doesn't allow for maintenance tasks like moving head, extrusion, etc.

closed 2019-09-07 by CoDanny