AstroPrint/OctoPrint-AstroPrint #22: Astro iOS app synchronization issue

AllenMcAfee, 2018-01-19:
### Bug Report
- Description: When synchronizing the printer view, the Extruder temperature reading updates the temperature intermittently.  Almost 50% of the time (after a successful sync), the extruder temperature section is blank.  The other 50% of the time it updates correctly.
- Expected behavior: The extruder temperature updates 100% of the time.
- Actual behavior: Almost 50% of the time (after a successful sync), the extruder temperature section is blank.  The other 50% of the time it updates correctly. The Bed Temperature is currently updating correctly 100% of the time.
- Steps to reproduce:
  - Connect to printer via the app
  - Click on temperature - updates 100% in this case
  - Allow phone to lock or go to sleep
  - Unlock phone, the app should load right away
  - Temperatures should update correctly, but issues above persist.

Hope this helps!

CoDanny, 2018-01-20:
This is a known bug of our iOS app.

CoDanny, 2018-03-19:
@AllenMcAfee Does this still happen with the latest version of the app 1.1.3

closed 2018-04-11 by CoDanny