AstroPrint/OctoPrint-AstroPrint #23: Something Went Wrong

3Dude10, 2018-01-30:
I have posted on here before but had to totally wipe my Pi.  Now I have a different issue, it says, "Installing the plugin from "https://github.com/AstroPrint/OctoPrint-AstroPrint/archive/master.zip" failed: Could not parse output from pip, see plugin_pluginmanager_console.log for generated output"  I am a newbie and not totally sure how this can be fixed, Thanks.

Alfonsopl, 2018-01-30:
Sorry but I could not reproduce the error.
Could you provide the plugin_pluginmanager_console.log data and check if it happens for all ways?
You can try to install the plugin by octoprint plugin managager (we are currently the second on the list), by providing our repository address or uploading the zip of master as you did.

Alfonsopl, 2018-01-31:
I was able to reproduce the error and apparently is a octoprint issue, it only happens when you try to install from url, until is fixed you can install it from plugin repository or from an uploaded archive. Thank you for the report.

3Dude10, 2018-02-02:
I tried both and I get the same message. Am I doing something else wrong?

Alfonsopl, 2018-02-02:
I´ll invistaigate and try to reproduce it.
Can you install other plugins or update octoprint?
What version  of octoprint  do you have?

3Dude10, 2018-02-03:
I have the newest octoprint, and I can install other plugins.

adam8797, 2018-02-20:
Ran into this same issue, was building OctoPrint on Ubuntu, not using OctoPi.

Seems to be a dependency issue. Running the following works:

```bash
$ sudo apt-get build-dep python-imaging
$ sudo apt-get install libjpeg8 libjpeg62-dev libfreetype6 libfreetype6-dev
```

closed 2019-08-11 by CoDanny