AstroPrint/OctoPrint-AstroPrint #25: How to add streaming video support

adam8797, 2018-02-20:
I know in the regular AstroBox release, you support streaming video via WebRTC. Is it possible to enable that for this plugin? Even if I need to manually install packages with apt-get, it would be nice to have a switch to enable it.

CoDanny, 2018-02-20:
It's a popular request. We need to take of a few other things first and then we'll look into this. You are right though, dependencies is our main fear and also the fact that we only test the video system in Raspbian ( where the AstroBox runs ) and OctoPrint can un in many more platforms.

rhys100, 2019-05-02:
What if you ONLY support rasp pi and just deny it to all other raspbian installs?

max5962, 2019-10-11:
@CoDanny 
Any news about that request ? :)
THnaks

CoDanny, 2019-10-11:
Hi @max5962 No news. This isn't something we can take on right now.

adam8797, 2019-10-11:
@CoDanny No worries. I’d appreciate if you leave this open, as this is something I’d love to see in the future.

CoDanny, 2019-10-12:
@adam8797 certainly we would like to have it also. Just need to clear out other priorities. Thanks for your understanding :)