AstroPrint/OctoPrint-AstroPrint #28: RASP Pi ZERO MK3 Loses access key / connection on Octoprint Plugin

rhys100, 2018-02-28:
Hi, I am not sure what variables are at play but I have the following Setup x2 and both have the same issue

RASP PI ZERO integrated into Prusa MK3 running 1.3.6 octoprint firmware and Astroprint Firmware.

Every day I need to goto the astroprint plugin tab and re input my api key and link them again.

I am not sure if the reset button on MK3 or other variables are at play but this is very annoying :)

Alfonsopl, 2018-03-01:
We use Oauth2  tokens for provide access to AstroPrint data and boxrouter. 

The only reason that  could make you get unlogged from the plugin is when you get your token expired/unauthorized. That happens after 16 days without using the plugin or very rarely when an error occurs in communications. It that happens you should get a warning in your logs with the message: "refresh token expired, AstroPrint user logged out".

If you can not find the message on this log it could because of the rasp pi zero, as  Lina indicates in octoprint installation: "Raspberry Pi Zero W is not recommended since severe performance issues were observed, caused by the WiFi interface when bandwidth is utilized (e.g. the webcam is streamed), negatively impacting printing results" so it that could be the reason.

Can you find the warning on your logs or try the plugin in a raspberry 2/3?

fred2088, 2018-03-07:
got the same problem even uninstall the plugin and install again ...

closed 2018-03-23 by Alfonsopl