AstroPrint/OctoPrint-AstroPrint #31: Ability to change printer name

luke321, 2018-03-14:
Hello,

I like this plugin an want to use it on all of my printers, however every printer added has the same name in the astroprint montor.
Is there a way to change this manually? Maybe you could add a name Input box in the astroprint plugin Tab.

Regards
Lukas

Alfonsopl, 2018-03-14:
We use box hostname for identify our boxes (and octoprint boxes too) by a name.

To change octopi hostname:
-Log in via SSH (pi@octopi.local password: raspberry, if you have more than 1 octoprint connected you will have to check its ip by your router and log by it: pi@167..... )
At the command line: 
- Sudo nano /etc/hostname and change the hostname from octopi. Don´t leave spaces before or after and save it.
- Sudo nano /etc/hosts, find the line with 127.0.1.1 , octopi and replace octopi with the same hostname
- Sudo reboot

So then you will be able to connect to you octopi by yournewname.local and see your box as yournewname.

closed 2018-03-14 by Alfonsopl

luke321, 2018-03-14:
Thank you for the fast response, maybe you could add that information to the Instructions.