AstroPrint/OctoPrint-AstroPrint #32: Printer name with multiple printers

TeeJS, 2018-03-20:
I have multiple OctoPrint-AstroPrint devices linked to my account.  It's difficult to use the monitor or print queue function of Astroprint, as all the printers are named "octopi"  could it use the Title from the appearance section of OctoPrint or give you an option to set the name in AstroPrint?
[image]

CoDanny, 2018-03-21:
@TeeJS We currently use your hostname. Which is given to you octopi at the system level, you can change it on the OS level. However we will look to see if we can facilitate this from our plugin.

TeeJS, 2018-03-21:
Knowing it takes it from the hostname is great, thanks!

closed 2018-03-23 by Alfonsopl