AstroPrint/OctoPrint-AstroPrint #53: No extruder fan on print

Watso4183, 2019-01-15:
The fan that points to the filament extruding to cool the job is not turning on when sending print jobs from Astroprint app, through this plugin. My device is a Monoprice Select v2. Please let me know what further information you might need to troubleshoot further.

CoDanny, 2019-01-15:
Please provide more information such us:

- Where is the print file sliced?
- Does the same print file works if started from OctoPrint?

Watso4183, 2019-01-20:
I've been using the Astroprint app, finding projects cia the thingiverse
add-on, and printing directly from the app. I will try pulling down on
gcode file and sending it via octoprint and will report back.