AstroPrint/OctoPrint-AstroPrint #55: Octoprint Backend with AstroPrint Touch

andrewsiemer, 2019-01-24:
I would love to be able to have the fully controllable octoprint as my backend web server while also having the AstroBox Touch as the front end on my enclosure!

CoDanny, 2019-01-25:
This isn't something we have plans on doing at the moment

closed 2019-01-25 by CoDanny

Deneteus, 2019-05-03:
Does this mean that if you install the plug-in that it is supposed to break your Octoprint install to where it only comes up with the touch interface? This was working prior to the release of Octoprint 0.16.0.

Which means I could use the plug-in to manage my downloads and print with no problems with .14 and .15 and I could get back to my octoprint web interface. Now it boots to nothing and SSH fails.

If so there needs to be some documentation changing on the plug-in listing via the Octoprint Plugin Repository.

https://plugins.octoprint.org/plugins/astroprint/

CoDanny, 2019-05-03:
@Deneteus I don't really know what you mean. Nothing has changed. The original poster wanted octoprint in the backend and AstroBox Touch UI in the screen. This isn't something we have ever supported or plan to support.

Deneteus, 2019-05-05:
> @Deneteus I don't really know what you mean. Nothing has changed. The original poster wanted octoprint in the backend and AstroBox Touch UI in the screen. This isn't something we have ever supported or plan to support.

'Does this mean that if you install the plug-in that it is supposed to break your Octoprint install to where it only comes up with the touch interface?'

That one sentence is exactly what I mean. After installing the Astroprint plug-in via the Octoprint Plugin installer my Octoprint Web interface no longer works and the Touch UI comes up via the PI HDMI. Now in the previous incarnation of the plug-in it didn't break my Octoprint Web interface and I could access the Astroprint features via a web browser. (Note I am not saying the touch interface was available, just that the web interface worked.)