AstroPrint/OctoPrint-AstroPrint #57: Wont Connect

sirprintsalot8, 2019-02-11:
I had the Astroprint Plugin working for about a month and then it stopped. I regenerated a code and input it and it wont see the printer and still shows the octoprint as authorized user back from the last date it work. I deleted both authorized user and readded them and the same thing. My android app goes to todays date when I redo it though. I then when into Octoprint and deleted the plugin and reinstalled. The results were the same. Sorry if I am missing something but I am at a loss.

CoDanny, 2019-02-11:
I'm lost as to how the Android app and the plugin are related, also what code was regenerated? and which authorized user? octoprint's or astroprint account?

Basically I'm not sure what the problem is by this description. Could you add some screenshots perhaps?

sirprintsalot8, 2019-02-12:
A full break down. On the Android app I cant see anything. So when I go to the Astroprint tab on Octoprint it just has to add an access key. That is the code I had regenerated and entered back in. But in the Astroprint page in the settings it has authorized users under the My Applications. My phone updates when I open the astroprint app but the octoprint plugin last access is still showing back in November when it stopped working.

[image]
[image]
[image]

CoDanny, 2019-02-14:
What happens when you enter the new access key and go through the auth flow?

sirprintsalot8, 2019-02-14:
I just regenerated the key and entered it. It goes to a page asking if I want to proceed. After I hit yes it refreshes and takes me back to the main Octoprint tab.

CoDanny, 2019-02-15:
Does it ask you the key again?

sirprintsalot8, 2019-02-15:
If I go back to the tab for the Astroprint plugin it looks like the screen shot from above.

CoDanny, 2019-02-17:
ugh! Ok. This is a problem that we have been chasing for a while but haven't been able to reproduce here. Could you post here a list of all the plugins you have installed and the version of OctoPrint? We'll try to reproduce your environment.

sirprintsalot8, 2019-02-18:
Plugins:

Action Command Prompt
AstroPrint
Bed Leveling Wizard
Bed Visualizer
Cancel Objects
Cost Estimation
Cost Plugin
CuraEngine (<= 15.04)
DeleteAfterPrint Plugin
DisplayLayerProgress Plugin
Fan Speed Control
GcodeEditor
Heater Timeout
MyMiniFactory
Navbar Temperature Plugin
OctoPrint Anywhere
Octolapse
PolarCloud
Print History Plugin
PrintTimeGenius Plugin
Printer Stats
Simple Emergency Stop
TP-Link Smartplug
Tab Order
Themeify
TouchUI

flyingelephant19, 2019-02-21:
i get the same issue. well everytime i shutdown the PI and restart, the creditials are gone and have to re enter and on one of my PI's i am running the latest 0.16 octopi.

CoDanny, 2019-08-11:
This should be resolved with #65

closed 2019-08-11 by CoDanny