AstroPrint/OctoPrint-AstroPrint #60: Octoprint Plugin cloud download fail

aplees, 2019-04-07:
The most recent update has caused an issue in octoprint. When selecting the cloud download to copy the file to pi local is sticks as 0% and fails to download. Have removed and reinstalled. Same issue. Updated my 2nd octoprint and now the same issue.

Alfonsopl, 2019-04-08:
I can not reproduce it, i have try different ways and looks fine.
Do you have the same problem when you try to print the print file instad download it, or when you send to print from astroprint?

aplees, 2019-04-09:
thanks for getting back to me.. i have managed to slice the file in astro print and move the file local to onto print no problem... the issue is downloading the STL within astroprint to the local SD card to do the slicing on the PI its self via the slicer plugin.. it hangs at 0% and never starts the download. have now also installed a fresh octoprint on the PI and installed the astro print plugin same failure.

CoDanny, 2019-08-11:
@aplees Have you tried this on version 1.3(2)

aplees, 2019-08-11:
Hi CoDanny. Sorry yes please close this. It fixed its self some time ago. Did go a few weeks with the issue and then one day it just worked.

closed 2019-08-11 by aplees