AstroPrint/OctoPrint-AstroPrint #62: Installing Astroprint Plug-in disables Octoprint Web UI, now boots Astroprint Touch Interface

Deneteus, 2019-05-05:
What is the problem?

Every time I install Astroprint plugin it kills my Octoprint server. After Astroprint installs the system never reboots. Manual reboot brings you to the white web page text that tells you to login and look at the logs to figure out what is wrong.

What did you already try to solve it?

I attempted to SSH into the server to identify what was going on and it won't even allow me to connect via SSH after it downloads the certificate.

When I went to connect an LCD to the PI to determine if I could get to the command line on a 7" lcd the screen was garbled. I then connected a 24" LCD monitor and I was greeted with the touch screen interface for the Astroprint. When I installed this before it did not do this so what am I missing here? It's supposed to be a plug-in. There was an Astroprint tab before and you could do everything from there and still see the Octoprint interface. What am I missing as I would have expected this from their download where you can add the 3.5" touch screen?

Note: I haven't been using the touch screen for anything on this particular PI since I used the web interface more than anything.

Additional information about your setup

Octoprint version: 0.16.0
Octopi version: current
Raspbian version: current
Printer: Prusa Mk3 / 3.7.0 firmware.
Hardware: Raspberry PI 3B+

CoDanny, 2019-05-05:
This is impossible as you describe it. The OctoPrint plugin has no code that puts anything on the HDMI interface. You might have inadvertently flashed the AstroBox Touch image on your pi.

Could you share some screenshots of both this: "I then connected a 24" LCD monitor and I was greeted with the touch screen interface for the Astroprint" and the file system on your sd card.

Deneteus, 2019-06-02:
> This is impossible as you describe it. The OctoPrint plugin has no code that puts anything on the HDMI interface. You might have inadvertently flashed the AstroBox Touch image on your pi.
>
> Could you [...]

I couldn't inadvertently flash anything. This was all done via the web interface the same way you would install any of the plug-ins. 

I actually took a screenshot on May 4th for reference. 
https://photos.app.goo.gl/EvBst24n1z168k7Q8

Deneteus, 2019-06-02:
I wiped the memory card out and reinstalled octoprint and everything is working.

closed 2019-06-02 by Deneteus