AstroPrint/OctoPrint-AstroPrint #70: self.astroprintCloud not set on printer status callback, endless stream of errors in the log

foosel, 2019-09-06:
I had to disable sentry.io logging for the current DSNs due to my instance getting swamped by an error produced within the printer status handler of this plugin:

```
AttributeError: 'NoneType' object has no attribute 'db'
  File "octoprint/printer/standard.py", line 196, in _sendCurrentDataCallbacks
    callback.on_printer_send_current_data(copy.deepcopy(data))
  File "octoprint_astroprint/printerlistener/__init__.py", line 93, in on_printer_send_current_data
    self.set_job_data(data['job'])
  File "octoprint_astroprint/printerlistener/__init__.py", line 120, in set_job_data
    cloudPrintFile = self.astroprintCloud.db.getPrintFileByOctoPrintPath(data['file']['path'])
```

Sentry ticket with stack traces and some context data [here](https://sentry.io/share/issue/76bf43e18a014971ac3cf645f4b6e3f6/). Until I turned things off I only saw two users or rather OctoPrint instances affected by this, together producing ~145k events within less than 4 days - it seems quite reproducible at least for these two. Since the sentry error tracking is anonymous, I cannot give you any contact data for further inquiries. I didn't see any events of the kind until September 2nd.

I'll patch OctoPrint's error tracking plugin with a filter for these errors in the next RC, it would probably be still a good idea to solve this on your end as well.

foosel, 2019-09-06:
After a quick look through your recent commits this *might* be solved by #69

CoDanny, 2019-09-06:
Thanks @foosel, we believe so too. This error happened as a side effect of a fix. The problem was quickly fixed in a subsequent release.

Thanks for reporting. Let us know if you see it pop up again

closed 2019-09-06 by CoDanny