AstroPrint/OctoPrint-AstroPrint #83: Plugin losing API key

greghesp, 2020-03-10:
[octoprint.log](https://github.com/AstroPrint/OctoPrint-AstroPrint/files/4308684/octoprint.log)
I think this may still be bugged. I checked my printer the other day to find no key, unsure if that was since the 1.4 update to octoprint though.

Regardless, I added my key back in but now it's saying it's locked?

Edit: just checked the plugin, key has been lost again. Will get the logs..

_Originally posted by @greghesp in https://github.com/AstroPrint/OctoPrint-AstroPrint/issues/79#issuecomment-596709853_

CoDanny, 2020-10-01:
@robertomilan logs please

adegard, 2020-10-16:
It happen to me each time I switch on Raspberry Pi (4B) and Octoprint... So pls find attached my log.
I'm newbee so no idea how it could help
[octoprint (adegard).log](https://github.com/AstroPrint/OctoPrint-AstroPrint/files/5393307/octoprint.adegard.log)

DaVinciQc, 2020-11-08:
Same issue here. Octoprint is losing API key every few minutes. It's impossible to use Astroprint.

[
[octoprint.log](https://github.com/AstroPrint/OctoPrint-AstroPrint/files/5506556/octoprint.log)
](url)