AstroPrint/OctoPrint-AstroPrint #84: Printer not getting the file from AstroPrint

tytech038, 2020-03-15:
My printer is connected to the Astroprint cloud. I tried to upload and slice in AstroPrint, That worked, but when i tried to send the print to the printer, It said it could not receive the file.
How do I fix this?

tytech038, 2020-03-15:
My octoprint.log fileoctoprint.log
[octoprint.log](https://github.com/AstroPrint/OctoPrint-AstroPrint/files/4335107/octoprint.log)

CoDanny, 2020-03-16:
The log doesn't show any relevant errors. This might be caused by network problems. In any case could you share a screenshot of the astroprint cloud's error?

tytech038, 2020-03-16:
This is what it says on my PC
[image]
And here is what it says on my phone (with CORS on)
[image]

CoDanny, 2020-03-16:
ok, weird. What's the status of the printer in the app? Not operational means that the raspberry pi is not connected to the printer. I'm assuming it is?

tytech038, 2020-03-18:
It is,
It prints just fine from TSD or Octoprint itself.

closed 2020-04-13 by tytech038