AstroPrint/OctoPrint-AstroPrint #89: Fetaure Enbales/Disable AstroPrint

Pomaroli, 2020-03-29:
it would be nice to have a function to enable/disable Astroprint from Octoprint plugin while not in use without to delete Astroprint account.

Alfonsopl, 2020-04-02:
I do not really understand the feature request .
You already can disable it by Octoprint Settings -> plugin manager.

[image]

If you disable it, and you enable it within 14 days, you will still be connected in your account.

Pomaroli, 2020-04-03:
yes true on the Astroprint Tab of octoprint the abiltity to register/unregstier